HenryHuYu/DiffPhysDrone #3: Install error

andrewjong, 2025-06-09:
Hi! I'm facing an error when trying to install the dependencies. Wondering if you might have any suggestions, please? Thank you.

```bash
# commands I ran
mamba create -n diffphysdrone python=3.11
mamba activate diffphysdrone
mamba install pytorch==2.2.2 torchvision==0.17.2 torchaudio==2.2.2 pytorch-cuda=11.8 -c pytorch -c nvidia
pip install -e src  # error here
```

Error output:

> Obtaining file:///home/andrew/Development/DiffPhysDrone/src
>   Preparing metadata (setup.py) ... done
> Installing collected packages: quadsim_cuda
>   DEPRECATION: Legacy editable install of quadsim_cuda==0.0.0 from file:///home/andrew/Development/DiffPhysDrone/src (setup.py develop) is deprecated. pip 25.3 will enforce this behaviour change. A possible replacement is to add a pyproject.toml or enable --use-pep517, and use setuptools >= 64. If the resulting installation is not behaving as expected, try using --config-settings editable_mode=compat. Please consult the setuptools documentation for more information. Discussion can be found at https://github.com/pypa/pip/issues/11457
>   Running setup.py develop for quadsim_cuda
>     error: subprocess-exited-with-error
>     
>     × python setup.py develop did not run successfully.
>     │ exit code: 1
>     ╰─> [85 lines of output]
>         running develop
>         /home/andrew/.miniforge3/envs/diffphysdrone/lib/python3.11/site-packages/setuptools/_distutils/cmd.py:90: DevelopDeprecationWarning: develop command is deprecated.
>         !!
>         
>                 ********************************************************************************
>                 Please avoid running ``setup.py`` and ``develop``.
>                 Instead, use standards-based tools like pip or uv.
>         
>                 By 2025-Oct-31, you need to update your project and remove deprecated calls
>                 or your builds will no longer be supported.
>         
>                 See https://github.com/pypa/setuptools/issues/917 for details.
>                 ********************************************************************************
>         
>         !!
>           self.initialize_options()
>         Obtaining file:///home/andrew/Development/DiffPhysDrone/src
>           Installing build dependencies: started
>           Installing build dependencies: finished with status 'done'
>           Checking if build backend supports build_editable: started
>           Checking if build backend supports build_editable: finished with status 'done'
>           Getting requirements to build editable: started
>           Getting requirements to build editable: finished with status 'error'
>           error: subprocess-exited-with-error
>         
>           × Getting requirements to build editable did not run successfully.
>           │ exit code: 1
>           ╰─> [23 lines of output]
>               Traceback (most recent call last):
>                 File "/home/andrew/.miniforge3/envs/diffphysdrone/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 389, in <module>
>                   main()
>                 File "/home/andrew/.miniforge3/envs/diffphysdrone/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 373, in main
>                   json_out["return_val"] = hook(**hook_input["kwargs"])
>                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>                 File "/home/andrew/.miniforge3/envs/diffphysdrone/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 157, in get_requires_for_build_editable
>                   return hook(config_settings)
>                          ^^^^^^^^^^^^^^^^^^^^^
>                 File "/tmp/pip-build-env-u9hjm7cp/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 473, in get_requires_for_build_editable
>                   return self.get_requires_for_build_wheel(config_settings)
>                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>                 File "/tmp/pip-build-env-u9hjm7cp/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 331, in get_requires_for_build_wheel
>                   return self._get_build_requires(config_settings, requirements=[])
>                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>                 File "/tmp/pip-build-env-u9hjm7cp/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 301, in _get_build_requires
>                   self.run_setup()
>                 File "/tmp/pip-build-env-u9hjm7cp/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 512, in run_setup
>                   super().run_setup(setup_script=setup_script)
>                 File "/tmp/pip-build-env-u9hjm7cp/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 317, in run_setup
[... 110 lines cut ...]
>         setup(
>       File "/home/andrew/.miniforge3/envs/diffphysdrone/lib/python3.11/site-packages/setuptools/__init__.py", line 115, in setup
>         return distutils.core.setup(**attrs)
>                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>       File "/home/andrew/.miniforge3/envs/diffphysdrone/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 186, in setup
>         return run_commands(dist)
>                ^^^^^^^^^^^^^^^^^^
>       File "/home/andrew/.miniforge3/envs/diffphysdrone/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 202, in run_commands
>         dist.run_commands()
>       File "/home/andrew/.miniforge3/envs/diffphysdrone/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 1002, in run_commands
>         self.run_command(cmd)
>       File "/home/andrew/.miniforge3/envs/diffphysdrone/lib/python3.11/site-packages/setuptools/dist.py", line 1102, in run_command
>         super().run_command(command)
>       File "/home/andrew/.miniforge3/envs/diffphysdrone/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 1021, in run_command
>         cmd_obj.run()
>       File "/home/andrew/.miniforge3/envs/diffphysdrone/lib/python3.11/site-packages/setuptools/command/develop.py", line 39, in run
>         subprocess.check_call(cmd)
>       File "/home/andrew/.miniforge3/envs/diffphysdrone/lib/python3.11/subprocess.py", line 413, in check_call
>         raise CalledProcessError(retcode, cmd)
>     subprocess.CalledProcessError: Command '['/home/andrew/.miniforge3/envs/diffphysdrone/bin/python3.11', '-m', 'pip', 'install', '-e', '.', '--use-pep517', '--no-deps']' returned non-zero exit status 1.
>     [end of output]
> 
> note: This error originates from a subprocess, and is likely not a problem with pip.
> 
>

andrewjong, 2025-06-09:
It seems to only occur from the `setup.py`. When I run `python -c "import torch; print(torch.__version__)"` it correctly prints 2.2.2.

HenryHuYu, 2025-06-15:
Sorry, have not encountered your problem. But typically it is simple cuda code and should be easy to compile. Some compatible problems need to modify .cu file with minor efforts. Maybe check your compiler and the cuda version of your torch via "nvcc -V " and "pip list | grep torch"

andrewjong, 2025-06-18:
@HenryHuYu thanks. To confirm, did you use a conda environment, or global pip? Also did you use a global cuda or a pip installed cuda?

Arclunar, 2025-06-18:
> [@HenryHuYu](https://github.com/HenryHuYu) thanks. To confirm, did you use a conda environment, or global pip? Also did you use a global cuda or a pip installed cuda?

I use a mamba environment，so is it  the cause ?

JiazheYuan, 2025-06-19:
> [@HenryHuYu](https://github.com/HenryHuYu) thanks. To confirm, did you use a conda/mamba environment, or global pip? Also did you use a global cuda or a pip installed cuda?

Just wanted to add what worked for me: I'm using a Conda environment with a global CUDA installation, and it's running successfully.

andrewjong, 2025-06-23:
@JiazheYuan May I ask what your Ubuntu version and global CUDA version are, please? Also, are you using miniforge conda, or anaconda? Thank you.