HenryHuYu/DiffPhysDrone #5: No drone in unity windows

Arclunar, 2025-06-23:
When running the code, Unity shows only the environment (no drone), but Gazebo and RViz display the drone correctly.

Could this be due to a configuration error on my part, or is the drone intentionally hidden in Unity? :(

[image]

[image]

[image]

MuFeng-mzyh, 2025-07-16:
Where can I find the visualization functionality in this project? In this open-source repository, I can only use main_cuda.py for training, but I'm unable to visualize the drone's operating environment. Could you briefly guide me on how to enable visualization? Thank you.

XXLiu-HNU, 2025-07-19:
> Where can I find the visualization functionality in this project? In this open-source repository, I can only use main_cuda.py for training, but I'm unable to visualize the drone's operating environment. Could you briefly guide me on how to enable visualization? Thank you.

You need to download the simulation validation code from the GitHub release page. To evaluate the trained model in multi-agent settings, use the following command to launch the simulator:
```
cd <path to multi agent code supplementary>
./LinuxNoEditor/Blocks.sh -ResX=896 -ResY=504 -windowed -WinX=512 -WinY=304 -settings=$PWD/settings.json
```

HenryHuYu, 2025-11-19:
> You need to download the simulation validation code from the GitHub release page. To evaluate the trained model in multi-agent settings, use the following command to launch the simulator:
>
> ```
> cd [...]

Thank you!