HenryHuYu/DiffPhysDrone #6: main_cuda.py segmentation fault

az40, 2025-07-01:
Environment Details:

Ubuntu 22.04
CUDA 12.8
PyTorch 2.7.1
Python 3.13.2

`python main_cuda.py $(cat configs/single_agent.args)` always segmentation faults at the same point (when training is 12% done).

Full output:
```
Namespace(resume=None, batch_size=64, num_iters=50000, coef_v=1.0, coef_speed=0.0, coef_v_pred=2.0, coef_collide=7.5, coef_obj_avoidance=3.0, coef_d_acc=0.01, coef_d_jerk=0.001, coef_d_snap=0.0, coef_ground_affinity=0.0, coef_bias=0.0, lr=0.001, grad_decay=0.4, speed_mtp=4.0, fov_x_half_tan=0.82, timesteps=150, cam_angle=20, single=True, gate=False, ground_voxels=True, scaffold=False, random_rotation=True, yaw_drift=True, no_odom=False)
  0%|                                                 | 0/50000 [00:00<?, ?it/s]/home/andrew/DiffPhysDrone/main_cuda.py:139: UserWarning: Using torch.cross without specifying the dim arg is deprecated.
Please either pass the dim explicitly or simply use torch.linalg.cross.
The default value of dim will change to agree with that of linalg.cross in a future release. (Triggered internally at /pytorch/aten/src/ATen/native/Cross.cpp:62.)
  R = torch.stack([fwd, torch.cross(up, fwd), up], -1)
loss: 2.605:  12%|██▋                    | 5881/50000 [12:33<1:34:35,  7.77it/s] Segmentation fault (core dumped)
```

Any ideas why this might be happening?

HenryHuYu, 2025-07-01:
could you try the recommand code 'simply use torch.linalg.cross'?

az40, 2025-07-02:
```
a = torch.randn(4, 3)
b = torch.randn(4, 3)
torch.linalg.cross(a, b)
```
executes with no problems

az40, 2025-07-07:
I don't know what the problem ended up being, but I tried setting it up on another device and ran training, and it was fine. Closing

closed 2025-07-07 by az40